hpsaturn/ESPNowCam #43: Guru Meditation Error with ILI9341

HeinzAlberto, 2024-05-29:
Hi there!

I try using an **ILI9341** with 240x320 Pixels over **SPI** with the basic reveiver example. I am using an ESP32 WROOM.
But whatever I do the ESP32 crashes with "Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled."

Maybe anyone here has an idea? :-D

So here's the code I am using:

````
#include <Arduino.h>
#include <ESPNowCam.h>
#include "spi_ili9341.h"
#include <LGFX_TFT_eSPI.hpp>

ESPNowCam radio;
LGFX tft;

#define LCD_CS 5

// frame buffer
uint8_t *fb; 
// display globals
int32_t dw, dh;

void onDataReady(uint32_t lenght) {
  tft.drawJpg(fb, lenght , 0, 0, dw, dh);
  // printFPS("MF:");
}

void setup() {
  Serial.begin(115200);

  pinMode(LCD_CS, OUTPUT);

  digitalWrite(LCD_CS, LOW);

  tft.init();
  tft.setRotation(1);
  tft.startWrite();

  dw = tft.width();
  dh = tft.height();

  if(psramFound()){
    size_t psram_size = esp_spiram_get_size() / 1048576;
    Serial.printf("PSRAM size: %dMb\r\n", psram_size);
  }

  // BE CAREFUL WITH IT, IF JPG LEVEL CHANGES, INCREASE IT
  fb = (uint8_t*)  ps_malloc(5000* sizeof( uint8_t ) ) ;

  radio.setRecvBuffer(fb);
  radio.setRecvCallback(onDataReady);

  if (radio.init()) {
    tft.setTextSize(2);
    tft.drawString("ESPNow Init success", dw / 6, dh / 2);
  }
  delay(1000);
}

void loop() {
}
````

And the content of **spi_ili9341.h**:
````
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device {
  lgfx::Panel_ILI9341 _panel_instance;
  lgfx::Bus_SPI _bus_instance;  // SPIバスのインスタンス

public:
  // コンストラクタを作成し、ここで各種設定を行います。
  // クラス名を変更した場合はコンストラクタも同じ名前を指定してください。
  LGFX(void) {
    {                                     // バス制御の設定を行います。
      auto cfg = _bus_instance.config();  // バス設定用の構造体を取得します。

      // SPIバスの設定
      cfg.spi_host = VSPI_HOST;  // 使用するSPIを選択  ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
      // ※ ESP-IDFバージョンアップに伴い、VSPI_HOST , HSPI_HOSTの記述は非推奨になるため、エラーが出る場合は代わりにSPI2_HOST , SPI3_HOSTを使用してください。
      cfg.spi_mode = 0;                   // SPI通信モードを設定 (0 ~ 3)
      cfg.freq_write = 40000000;          // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)
      cfg.freq_read = 16000000;           // 受信時のSPIクロック
      cfg.spi_3wire = true;               // 受信をMOSIピンで行う場合はtrueを設定
      cfg.use_lock = true;                // トランザクションロックを使用する場合はtrueを設定
      cfg.dma_channel = SPI_DMA_CH_AUTO;  // 使用するDMAチャンネルを設定 (0=DMA不使用 / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=自動設定)
      // ※ ESP-IDFバージョンアップに伴い、DMAチャンネルはSPI_DMA_CH_AUTO(自動設定)が推奨になりました。1ch,2chの指定は非推奨になります。
      cfg.pin_sclk = 18;  // SPIのSCLKピン番号を設定
      cfg.pin_mosi = 23;  // SPIのMOSIピン番号を設定
      cfg.pin_miso = -1;  // SPIのMISOピン番号を設定 (-1 = disable)
      cfg.pin_dc = 17;    // SPI DC-Pin

      _bus_instance.config(cfg);               // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);  // バスをパネルにセットします。
    }

    {                                       // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config();  // 表示パネル設定用の構造体を取得します。

      cfg.pin_cs = -1;    // CS要拉低 -1
      cfg.pin_rst = -1;   // RST和开发板RST相连 -1
      cfg.pin_busy = -1;  // BUSYが接続されているピン番号 (-1 = disable)

      // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

      //cfg.memory_width = 240;                   // ドライバICがサポートしている最大の幅
      //cfg.memory_height = 320;                  // ドライバICがサポートしている最大の高さ
      cfg.panel_width = 240;     // 実際に表示可能な幅
      cfg.panel_height = 320;    // 実際に表示可能な高さ
      cfg.offset_x = 0;          // パネルのX方向オフセット量
      cfg.offset_y = 0;          // パネルのY方向オフセット量
      cfg.offset_rotation = 0;   // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel = 8;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits = 1;   // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable = true;       // データ読出しが可能な場合 trueに設定
      cfg.invert = false;        // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order = false;     // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit = false;    // データ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared = false;    // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance);  // 使用するパネルをセットします。
  }
};

````

HeinzAlberto, 2024-05-29:
**Update:**
The ESP32 only crashes when it recieves data from the ESONowCam-Sender - at least I think so. If the sender is plugged out, everything's fine.

hpsaturn, 2024-05-29:
Are you using what board? Maybe is possibly that you paste here the output of the exception? and try to launch it with PlatformIO, because it has a exception decoder. For that please test with this sample that should work with the 9341

```bash
pio run -e tft-il9485-basic-receiver --target upload
```

HeinzAlberto, 2024-05-29:
I wasn't able to debug the ESP32 with PlatformIO and was not able to decode the exception - but increasing the malloc allocation and changing `ps_malloc` to `malloc` did the trick! - It does not crash anymore.

Seems like the only problem left is the non-working screen, but this could also be a problem of my LGFX-configuration...

hpsaturn, 2024-05-31:
ok good. Thanks for the feedback. Please let me know if you resolve the screen issue to try to improve the examples.. or please submit a pull request with the respective example using your screen.

closed 2024-05-31 by hpsaturn

hpsaturn, 2024-05-31:
A friend of mine shared that LGFX has auto detection. Maybe is a good alternative for your issue:
https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/1_simple_use/1_simple_use.ino

HeinzAlberto, 2024-06-01:
Thanks! I tried this as well but my board and screen seems not to be covered by the autodetection.

I managed to build an new example which uses another GFX-Library (Adafruit_GFX). Since this library works for my screen but is not able to process jpg, I used an jpg-to-bmp conversion. Speed is "okay", the main problem now seems to be the speed of the SPI bus. Anyway I will submit a pull request with the example ^_^