hpsaturn/ESPNowCam #50: Why PIXFORMAT_RGB565 instead of PIXFORMAT_JPEG

RickLloyd24, 2024-12-19:
This is not an issue, it just a curiosity question.   I am using the CamAIThinker and the CamAIThinker.h file.  The camera "config.pixel_format  = PIXFORMAT_RGB565 is used".  Then the RG565 is converted to jpg with "frame2jpg(Camera.fb, 12, &out_jpg, &out_jpg_len)".  Why didn't you set camera config.pixel_format  = PIXFORMAT__JPEG ?

Sorry if this is a stupid question.

hpsaturn, 2024-12-19:
Don't worries, the question is valid. Please check in the examples directory:
https://github.com/hpsaturn/ESPNowCam/blob/master/examples/freenove-nojpg-sender/freenove-nojpg-sender.cpp
For instance there you are able to use it in this way.

Also, I'm working in an utility to configure all cameras with a CLI running in the ESP32:
https://github.com/hpsaturn/esp32-camera-cli/tree/master
There you can see that is easy toggle into the both cases.

hpsaturn, 2024-12-19:
I forgot say that it was a question also for me when I started with this library. Exists a difference to use the external JPG library from Arduino Framework versus the internal implementation of JPEG on the IDF Camera driver. With my esp32-camera-cli, you can play with these settings and note this difference.

hpsaturn, 2025-01-27:
Closed it because it was without any response. Seems that is resolved.

closed 2025-01-27 by hpsaturn