hpsaturn/ESPNowCam #51: Error when building an image

phpo1-61, 2024-12-27:
Build esp32cam-p2p-sender on Fedora 41 Linux Workstation (kernel 6.12.
.4-200.fc41.x86_64)

VSCode version: 1.96.0
Date: 2024-12-11 1T02:29:09.626Z(2wks ago)
Version: 1.96.0
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Linux x64 6.12.4-200.fc41.x86_64

platformIO version:

admin@pc-29:~/Documents/PlatformIO/Projects/test1$ pio system info
--------------------------  --------------------------------------------------
PlatformIO Core             6.1.16
Python                      3.13.0-final.0
System Type                 linux_x86_64
Platform                    Linux-6.12.4-200.fc41.x86_64-x86_64-with-glibc2.40
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/admin/.platformio
PlatformIO Core Executable  /home/admin/.platformio/penv/bin/platformio
Python Executable           /home/admin/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10

platformio.ini file: file supplied in the ESP Now Cam 0.0.16 package

Pio Console command:
 pio run -e esp32cam-p2p-sender --target uoload

error message:
*** missing SConscript file 'prebuild.py'
File "/home/admin/.platformio/penv/lib64/python3.13/site-packages/platformio/builder/main.py", line 167, in <module>

refers to line:

env.SConscript(env.GetExtraScripts("pre"), exports="env")

How can I fix this error?
Thank you for jour help.

hpsaturn, 2024-12-28:
Hi, well, first please check some things:

1. check your PlatformIO installation
2. check the upload option in your command line, you wrote "uoload"
3. run `pio pkg update`
4. run `pio upgrade`
5. run again: `pio run -e esp32cam-p2p-sender` only, for test only the build process.
6. run with `upload`

On the other hand, the continuous integration system test are passing ok:
https://github.com/hpsaturn/ESPNowCam/actions/runs/12502843155/job/34882348516

Please let me know if you fix your issue.

phpo1-61, 2024-12-29:
Hi, Thanks for your answer: I am nearly a beginner with VSCode and PlatformIO 
wich is much more complicated to me than Arduino IDE and I probably made some 
errors.

In fact,
I performed the actions you mentionned in your answer:

from the root directory of my project and in the Core pltformIO console:
pio pkg update
pio pkg upgrade

result: my configuration is up to date.

In fact, I have created a new project "from srcatch" in order not to be 
disturbed with the two previous where I probably made many mistakes, however 
it was a good method to become familiar with VsCode and PlatformIO.

This project is named test3 and deals with your file esp32cam-p2p-sender that I 
copied in the main.cpp file of PLatformIO.
I reduced the platform.ini file to the following statements:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32cam]
platform = espressif32
framework = arduino
monitor_speed = 115200
monitor_filters =
 esp32_exception_decoder
 time
build_flags =
  ; -Wall
  ; -Wextra
  ; -Werror
  -D CORE_DEBUG_LEVEL=3
  -D BOARD_HAS_PSRAM=1
lib_deps = hpsaturn/EspNowCam@^0.1.16
; extra_scripts = pre:prebuild.py

I have commented with a semi-column the last line of the [env:espcam32] 
section in order to fix the error message I reported in my issue (*** missing 
SConscript file 'prebuild.py') and now compilation and upload are successful.

Is this statement extra_script compulsory for camera operation? I just red 
that in PlatformIO web site that it was intended for experienced programmers, 
out of them I don't belong.

Once this first step passed, I want to test camera operation with ESPNoCam and 
a receiver on a TFT screen.I already own an AI Thinker ESPCam and a LILYGO 
ESP32 T-Display equipped with a TFT 1.14 inch TFT display (240 X 135 pixels). 
My first question deals with the compatibility of this receiver screen pixel 
configuration with the OV2640 device of AI Thinker Cam? If not, I will purchase 
one of the ESP32 TFT screen already tested.

Thank you anyway for your kind help.
Best regards.

Philippe Poutrel.

Le samedi 28 décembre 2024, 12:51:11 heure normale d’Europe centrale hpsaturn 
a écrit :
> Hi, well, first please check some things:
> 
> 1. check your PlatformIO installation
> 2. check the upload option in your command line, you wrote "uoload"
> 3. run `pio pkg update`
> 4. run `pio upgrade`
> 5. run again: `pio run -e esp32cam-p2p-sender` only, for test only the build
> process. 6. run with `upload`
> 
> On the other hand, the continuous integration system test are passing ok:
> https://github.com/hpsaturn/ESPNowCam/actions/runs/12502843155/job/348823485
> 16
> 
> Please let me know if you fix your issue.

hpsaturn, 2024-12-30:
> Is this statement extra_script compulsory for camera operation? 

Nop. This PlatformIO ini file in the library repo is only for run the examples and tests from the up level of this directory. Like you did is the right way. A new and reduced ini file is a better idea. On the other hand, the examples should run like is described in the README file. I wondering why not in your case. In my environment and also in GitHub the examples are running ok. Whatever, a isolated project is better for understand better all here. Go head with your choose. 

> Once this first step passed, I want to test camera operation with ESPNoCam and a receiver on a TFT screen.I already own an AI Thinker ESPCam and a LILYGO ESP32 T-Display equipped with a TFT 1.14 [...]

Well, I remember that I did also tests with the same TFT display and it works with this resolution.. In theory any display should work like receptor. But is more easy first, test the TFT display with LovyanGFX Library, because all examples and tests that I did worked with this library. But should works with any library that has support to write JPG frames.

phpo1-61, 2025-01-05:
Le dimanche 29 décembre 2024 22:52:30 heure normale d’Europe centrale, vous 
avez écrit :
> Hi, Thanks for your answer: I am nearly a beginner with VSCode and
> PlatformIO wich is much more complicated to me than Arduino IDE and I
> probably made some errors.
> 
> In fact,
> I performed the actions you mentionned in your answer:
> 
> from the root directory of my project and in the Core pltformIO console:
> pio pkg update
> pio pkg upgrade
> 
> result: my configuration is up to date.
> 
> In fact, I have created a new project "from srcatch" in order not to be
> disturbed with the two previous where I probably made many mistakes, however
> it was a good method to become familiar with VsCode and PlatformIO.
> 
> 
> 
> This project is named test3 and deals with your file esp32cam-p2p-sender
> that I copied in the main.cpp file of PLatformIO.
> I reduced the platform.ini file to the following statements:
> 
> ; PlatformIO Project Configuration File
> ;
> ;   Build options: build flags, source filter
> ;   Upload options: custom upload port, speed and extra flags
> ;   Library options: dependencies, extra library storages
> ;   Advanced options: extra scripting
> ;
> ; Please visit documentation for the other options and examples
> ; https://docs.platformio.org/page/projectconf.html
> 
> [env:esp32cam]
> platform = espressif32
> framework = arduino
> monitor_speed = 115200
> monitor_filters =
>  esp32_exception_decoder
>  time
> build_flags =
>   ; -Wall
>   ; -Wextra
>   ; -Werror
>   -D CORE_DEBUG_LEVEL=3
>   -D BOARD_HAS_PSRAM=1
> lib_deps = hpsaturn/EspNowCam@^0.1.16
> ; extra_scripts = pre:prebuild.py
> 
> I have commented with a semi-column the last line of the [env:espcam32]
> section in order to fix the error message I reported in my issue (***
> missing SConscript file 'prebuild.py') and now compilation and upload are
> successful.
> 
> Is this statement extra_script compulsory for camera operation? I just red
> that in PlatformIO web site that it was intended for experienced
> programmers, out of them I don't belong.
> 
> Once this first step passed, I want to test camera operation with ESPNoCam
> and a receiver on a TFT screen.I already own an AI Thinker ESPCam and a
> LILYGO ESP32 T-Display equipped with a TFT 1.14 inch TFT display (240 X 135
> pixels). My first question deals with the compatibility of this receiver
> screen pixel configuration with the OV2640 device of AI Thinker Cam? If
> not, I will purchase one of the ESP32 TFT screen already tested.
> 
> Thank you anyway for your kind help.
> Best regards.
> 
[... 246 lines cut ...]
  Serial.println("setup ends");

  dw = tft.width();
  dh = tft.height();
  Serial.print("acquired screen width: ");
  Serial.println(dw);
  Serial.print("acquired screen height: ");
  Serial.println(dh);
  if(psramFound()){
    size_t psram_size = esp_spiram_get_size() / 1048576;
    Serial.printf("PSRAM size: %dMb\r\n", psram_size);
  }

  // BE CAREFUL WITH IT, IF JPG LEVEL CHANGES, INCREASE IT
  fb = (uint8_t*)  malloc(5000* sizeof( uint8_t ) ) ;

  radio.setRecvBuffer(fb);
  radio.setRecvCallback(onDataReady);

  if (radio.init()) {
    tft.setTextSize(2);
    tft.drawString("ESPNow Init success", dw / 6, dh / 2);
  }
  delay(1000);
}

hpsaturn, 2025-01-05:
Hi, thanks for this complete feedback. I'm glad to know that is working, and thanks to share de files for that, maybe I could include it to the examples, the code for the T-Display. I'm going to close this issue for now, because all doubts was resolved.

closed 2025-01-05 by hpsaturn

phpo1-61, 2025-01-13:
Le dimanche 5 janvier 2025, 23:59:16 heure normale d’Europe centrale hpsaturn 
a écrit :
> Closed #51 as completed.

Hi,

I successfully tested ESPNowCam with an ESP32cam AIThinker and a Lilygo TTGO 
T-Display.
First I had a dark and negative picture on the TFT display, so after looking 
for some ideas on websites, I added lines on the main.cpp file in order to 
adjust the camera's parameters, especially set_special_effects and set_wb_mode. 
And the result is quite satisfactory.

You fill find hereafter the whole configuration of my tests:

IDE and platform:

VSCode version: 1.96.0
Date: 2024-12-11 1T02:29:09.626Z(2wks ago)
Version: 1.96.0
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Linux x64 6.12.4-200.fc41.x86_64

platformIO version:

***@***.***:~/Documents/PlatformIO/Projects/test1$ pio system info
--------------------------  --------------------------------------------------
PlatformIO Core             6.1.16
Python                      3.13.0-final.0
System Type                 linux_x86_64
Platform                    Linux-6.12.4-200.fc41.x86_64-x86_64-with-glibc2.40
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/admin/.platformio
PlatformIO Core Executable  /home/admin/.platformio/penv/bin/platformio
Python Executable           /home/admin/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10

Platform used for PLatformIO: Arduino

ESP32Cam AIThinker configuration:

platformio.ini:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
monitor_speed = 115200
monitor_filters =
  esp32_exception_decoder
  time
[... 292 lines cut ...]
  delay(1000);
}

void loop() {
}

I finally performed a last trial: using DRAM instead of PSRAM on AIThinker card 
by changing these lines in CamAIThinker.h file:

config.pixel.format changed from PIXFORMAT_RGB565 to PIXFORMAT_JPEG
and:
fb_location from CAMERA_FB_IN_PSRAM to CAMERA_FB_IN_DRAM

and commenting the BOARD_HAS_PSRAM in platformio.ini file.

Unfortunately it led to a failure: just colored vertical strips were displayed 
on the screen and despite the activation of a build_type = debug commande in 
the receiver card platformio.ini file, I got no debug information.

Anyway I am satisfied with this project and plan to purchase a larger screen 
among those you have tested to improve user's comfort.

Best regards.

Philippe Poutrel.

hpsaturn, 2025-01-13:
For improve the debug level, or have more messages, you should change this flag:
```python
build_flags = 
  -D CORE_DEBUG_LEVEL=0         # For debugging set to 3 and enable debug mode in the app
``` 
One tip more:
Also I recommend you please read how to write in Markdown syntax, for in a future have better comments in this threads in GitHub, for instance for show your code like this: