hpsaturn/ESPNowCam #53: Core exception with ESP32-8048S043 board used as ESPNowCam receiver

phpo1-61, 2025-01-27:
As receiver of ESPNowCam protocol flow, I am testing the ESP32-8048S043
board (ESP32 S3 WROOM-1 with capacity touch 4,3 inch 800 X 400 LCD TFT
RGB screen, ILI9485 driver according to the Epressif datasheet for this
processor). I choosed this model because il matches the driver file
parameters named LGFX_ESP32S3_RGB_ESP32-8048S043.h in github ESPNowcam
website.

I suppose that this processor is not fitted with internal PSRAM,
otherwise its denomination would involve the "R2" or "R8" pattern and no
other clue shows that a PSRAM is included on the board. It seems to be
confirmed by the fact that the board used in the
tft-rgb-basic-receiver.cpp example file (github ESPNowCam website) is
esp32-s3-devkitc-1 and appears with the attributes N8(8MB-QD, No PSRAM)
in PLatformIO when choosing a board, new project creation process.

My work environment is: is:

SCode version: 1.96.0 Date: 2024-12-11 1T02:29:09.626Z(2wks ago)
Version: 1.96.0 Commit: 138f619c86f1199955d53b4166bef66ef252935c Date:
2024-12-11T02:29:09.626Z Electron: 32.2.6 ElectronBuildId: 10629634
Chromium: 128.0.6613.186 Node.js: 20.18.1 V8: 12.8.374.38-electron.0 OS:
Linux x64 6.12.4-200.fc41.x86_64

platformIO version:

admin@pc-29:~/Documents/PlatformIO/Projects/test1$ pio system info

------------------------------------------------------------------------

PlatformIO Core 6.1.16 Python 3.13.0-final.0 System Type linux_x86_64
Platform Linux-6.12.4-200.fc41.x86_64-x86_64-with-glibc2.40 File System
Encoding utf-8 Locale Encoding UTF-8 PlatformIO Core Directory
/home/admin/.platformio PlatformIO Core Executable
/home/admin/.platformio/penv/bin/platformio Python Executable
/home/admin/.platformio/penv/bin/python Global Libraries 0 Development
Platforms 1 Tools & Toolchains 10

I use the Arduino platform within PLatformIO, not ESP-IDF.

First, I made a simple configuration in order to test the display
function based on the tft-rgb-basic-receiver.cpp and platformio.ini just
to check the completion of basic Serial.print and fillscreen TFT_RED
commands.

In main.cpp file:

-   all commands dealing with ESPNoCam library have been commented
-   #DEFINE CORE_DEBUG-LEVEL =3 has been added (in order to get detailed
    ESP32 exceptions messages)

In platformIO.ini file:

-   board_build.flash_size = 16MB command has been commented because my
    flash size is 8Mb, just like the path and name of the partitions.csv
    file
-   -D BOARD_HAS_PSRAM has been commented, as suggested previously
-   only Lovyan03 library is loaded in working environment

platformio.ini file:

; PlatformIO Project Configuration File ; ; Build options: build flags,
source filter ; Upload options: custom upload port, speed and extra
flags ; Library options: dependencies, extra library storages ; Advanced
options: extra scripting ; ; Please visit documentation for the other
options and examples ; https://docs.platformio.org/page/projectconf.html

\[env:esp32-s3-devkitc-1\] platform = espressif32 board =
esp32-s3-devkitc-1 ; board_build.flash_size = 16MB ;
board_build.partition =
"home/admin/Documents/PlatformIO/Projects/test9/.pio/libdeps/esp32-s3-devkitc-1/EspNowCam/config/partitions.csv"board_build.flash_size
[... 318 lines cut ...]
22:56:20.280 \> 22:56:20.280 \> 22:56:20.411 \> 22:56:20.411 \>
22:56:20.411 \> 22:56:20.411 \> ELF file SHA256: 556f4b65449fb844
22:56:20.411 \>

At last, I achieved 2 extra tests:

-   with D BOARD_HAS_PSRAM in platformio.ini and 1.2.0 release
    (Lovyan03): compilation blocking error message:

PSRAM chip not found or bad type or wrong PSRAM line mode.

If at last PSRAM actually exists on this board, how could I declare it
with DIO mode in platformio.ini file (this information is given in
Exception messages) since I understood that FLASH and PSARM operates the
same SPI transfert format mode (SPI, Dual SPI, Quad SPI, OctalSPI, QPI,
OPI)

-   Without D BOARD_HAS_PSRAM, with board_build.flash_size = 16MB (and
    partitions.csv file entered) and 1.2.0 Lovyan03 library relase :
    same behavior and exception messages

I would like to have your advice in order to investigate about the
origin of this issue because I'm now running out of ideas.

Best regards.

hpsaturn, 2025-01-27:
Thanks for your comments and extended report, but is it possible that you improve the readability of this using Markdown blocks please? for understand better and with that I can help you. 

https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

phpo1-61, 2025-01-27:
Le lundi 27 janvier 2025, 21:35:48 heure normale d’Europe centrale hpsaturn a 
écrit :
> Thanks for your comments and extended report, but is it possible that you
> improve the readability of this using Markdown blocks please? for
> understand better and with that I can help you.
> 
> https://docs.github.com/en/get-started/writing-on-github/working-with-advanc
> ed-formatting/creating-and-highlighting-code-blocks
OK, I am going to work with pandoc tool in order to produce a markdown 
document. 
I think that my issue is in touch with memory handling. On an example 
configured for this boad with Arduino IDE, flash mode is set to QIO 80Mhz 
whereas core debug messages tell about DIO.

Regards.

hpsaturn, 2025-01-27:
Well, here is enough, you don't need a external tool. Just type the blocks like is described in this guide. It is very easy. For instance for a C++ block you should start with triple single quote, like this:

[image]

And this will generate this:

```cpp
Serial.begin(115200);

delay(1000); // only for debugging 

if(psramFound()){
    size_t psram_size = esp_spiram_get_size() / 1048576;
    Serial.printf("PSRAM size: %dMb\r\n", psram_size);
}
```

phpo1-61, 2025-01-28:
Le lundi 27 janvier 2025, 23:27:57 heure normale d’Europe centrale hpsaturn a 
écrit :
> Well, here is enough, you don't need a external tool. Just type the blocks
> like is described in this guide. It is very easy. For instance for a C++
> block you should start with triple single quote, [...]
Hi, you will find hereafter the text embedded with triple single quote for C++ 
blocks, as for other messages blocks.

Best regards.

As receiver of ESPNowCam protocol flow, I am testing the ESP32-8048S043 board 
(ESP32 S3 WROOM-1 with capacity touch 4,3 inch 800 X 400 LCD TFT RGB screen, 
ILI9485 driver according to the Epressif datasheet for this processor). I 
choosed this model because il matches the driver file parameters named 
LGFX_ESP32S3_RGB_ESP32-8048S043.h in github ESPNowcam website.

I suppose that this processor is not fitted with internal PSRAM, otherwise its 
denomination would involve the "R2" or "R8" pattern and no other clue shows 
that a PSRAM is included on the board. It seems to be confirmed by the fact 
that the board used in the tft-rgb-basic-receiver.cpp example file (github 
ESPNowCam website) is esp32-s3-devkitc-1 and appears with the attributes 
N8(8MB-QD, No PSRAM) in PLatformIO when choosing a board, new project creation 
process.

My work environment is: is:

SCode version: 1.96.0
Date: 2024-12-11 1T02:29:09.626Z(2wks ago)
Version: 1.96.0
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Linux x64 6.12.4-200.fc41.x86_64

platformIO version:

***@***.***:~/Documents/PlatformIO/Projects/test1$ pio system info
--------------------------  --------------------------------------------------
PlatformIO Core             6.1.16
Python                      3.13.0-final.0
System Type                 linux_x86_64
Platform                    Linux-6.12.4-200.fc41.x86_64-x86_64-with-glibc2.40
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/admin/.platformio
PlatformIO Core Executable  /home/admin/.platformio/penv/bin/platformio
Python Executable           /home/admin/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10

I use the Arduino platform within PLatformIO, not ESP-IDF.

First, I made a simple configuration in order to test the display function 
based on the tft-rgb-basic-receiver.cpp and platformio.ini just to check the 
completion of basic Serial.print  and  fillscreen TFT_RED commands.

In main.cpp file:
- all commands dealing with ESPNoCam library have been commented
- #DEFINE CORE_DEBUG-LEVEL =3  has been added (in order to get detailed ESP32 
exceptions messages)

In platformIO.ini file:
  - board_build.flash_size = 16MB command has been validated atlthough I'm not 
shure it is the exact, as path lib to partitions.csv file
 - -D BOARD_HAS_PSRAM has been commented, as suggested previously
[... 454 lines cut ...]
22:56:20.411 >
22:56:20.411 > ELF file SHA256: 556f4b65449fb844
22:56:20.411 >
'''

At last, I achieved 2 extra tests:

 - with D BOARD_HAS_PSRAM in platformio.ini and 1.2.0 release (Lovyan03): 
compilation blocking error message:

PSRAM chip not found or bad type or wrong PSRAM line mode.

If at last PSRAM actually exists on this board, how could I declare it with 
DIO mode in platformio.ini file (this information is given in Exception 
messages) since I understood that FLASH and PSARM operates the same SPI 
transfert format mode (SPI, Dual SPI, Quad SPI, OctalSPI, QPI, OPI)

 - Without D BOARD_HAS_PSRAM, nor board_build.flash_size = 16MB (and 
partitions.csv file entered) and 1.2.0 Lovyan03 library relase : same behavior 
and exception messages as previous trials

I would like to have your advice in order to investigate about the origin of 
this issue because I'm now running out of ideas.

Best regards.

hpsaturn, 2025-01-28:
Nice try, but not, is with a different `
You should use this `
Three for opening, Three for close each block
Please see some posts in this thread, we used these blocks:
https://github.com/hpsaturn/ESPNowCam/issues/10

phpo1-61, 2025-01-28:
Le mardi 28 janvier 2025, 15:53:46 heure normale d’Europe centrale hpsaturn a 
écrit :
> `
Sorry, Is this better with this new triple quote?

As receiver of ESPNowCam protocol flow, I am testing the ESP32-8048S043 board 
(ESP32 S3 WROOM-1 with capacity touch 4,3 inch 800 X 400 LCD TFT RGB screen, 
ILI9485 driver according to the Epressif datasheet for this processor). I 
choosed this model because il matches the driver file parameters named 
LGFX_ESP32S3_RGB_ESP32-8048S043.h in github ESPNowcam website.

I suppose that this processor is not fitted with internal PSRAM, otherwise its 
denomination would involve the "R2" or "R8" pattern and no other clue shows 
that a PSRAM is included on the board. It seems to be confirmed by the fact 
that the board used in the tft-rgb-basic-receiver.cpp example file (github 
ESPNowCam website) is esp32-s3-devkitc-1 and appears with the attributes 
N8(8MB-QD, No PSRAM) in PLatformIO when choosing a board, new project creation 
process.

My work environment is:
```
SCode version: 1.96.0
Date: 2024-12-11 1T02:29:09.626Z(2wks ago)
Version: 1.96.0
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Linux x64 6.12.4-200.fc41.x86_64

platformIO version:

***@***.***:~/Documents/PlatformIO/Projects/test1$ pio system info
--------------------------  --------------------------------------------------
PlatformIO Core             6.1.16
Python                      3.13.0-final.0
System Type                 linux_x86_64
Platform                    Linux-6.12.4-200.fc41.x86_64-x86_64-with-glibc2.40
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/admin/.platformio
PlatformIO Core Executable  /home/admin/.platformio/penv/bin/platformio
Python Executable           /home/admin/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10

I use the Arduino platform within PLatformIO, not ESP-IDF.
```

First, I made a simple configuration in order to test the display function 
based on the tft-rgb-basic-receiver.cpp and platformio.ini just to check the 
completion of basic Serial.print  and  fillscreen TFT_RED commands.

In main.cpp file:
- all commands dealing with ESPNoCam library have been commented
- #DEFINE CORE_DEBUG-LEVEL =3  has been added (in order to get detailed ESP32 
exceptions messages)

In platformIO.ini file:
 - board_build.flash_size = 16MB command has been used, although I am not sure 
of this size
 - -D BOARD_HAS_PSRAM has been commented, as suggested previously
 - only Lovyan03 library is loaded in working environment

 platformio.ini file:
```
[... 449 lines cut ...]
22:56:20.411 >
22:56:20.411 > ELF file SHA256: 556f4b65449fb844
22:56:20.411 >
```

At last, I achieved 2 extra tests:

 - with D BOARD_HAS_PSRAM in platformio.ini and 1.2.0 release (Lovyan03): 
compilation blocking error message:

PSRAM chip not found or bad type or wrong PSRAM line mode.

If at last PSRAM actually exists on this board, how could I declare it with 
DIO mode in platformio.ini file (this information is given in Exception 
messages) since I understood that FLASH and PSARM operates the same SPI 
transfert format mode (SPI, Dual SPI, Quad SPI, OctalSPI, QPI, OPI)

 - Without D BOARD_HAS_PSRAM, with board_build.flash_size = 16MB, without 
partitions.csv file entered, 1.2.0 Lovyan03 library release loaded: same 
behavior and exception messages

I would like to have your advice in order to investigate about the origin of 
this issue because I'm now running out of ideas.

Best regards.

hpsaturn, 2025-01-28:
sorry, your block codes are not working, maybe because you are using bad quotes, or you not closed these blocks..

The block codes should show like this:

```python
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.flash_size = 16MB
board_build.partition = "home/admin/Documents/PlatformIO/Projects/test14/.pio/
libdeps/esp32-s3-devkitc-1/EspNowCam/config/partitions.csv"
monitor_speed = 115200
monitor_filters =
  esp32_exception_decoder
  time
build_flags =
  ; -Wall
  ; -Wextra
  ; -Werrors
  ;-D CORE_DEBUG_LEVEL= 3
  ;-D BOARD_HAS_PSRAM
  ; Log2File
build_type = debug
framework = arduino
lib_deps =
	; hpsaturn/EspNowCam@^0.1.16
	lovyan03/LovyanGFX@^1.2.0
```

Can you see this? I'm going to share a image for you, for understand how to it should showed:

[image]

phpo1-61, 2025-01-28:
Le mardi 28 janvier 2025, 20:27:32 heure normale d’Europe centrale hpsaturn a 
écrit :
> sorry, your block codes are not working, maybe because you are using bad
> quotes, or you not closed these blocks..
>
> The block codes should show like this:
>
> ```python
> [env:esp32-s3-devkitc-1]
> platform [...]
let's begin with main.cpp (platformio.ini file is OK in your image). 
Is marking correct hereafter?

```
/**
 * Sample shared by @MeloCuentan
 * Board:
 * https://www.aliexpress.com/item/1005006625303218.html
 */

#include <Arduino.h>
#include <HardwareSerial.h>
// #include "home/admin/Documents/PLatformIO/Projects/test14/.pio/libdeps/
esp32-s3_devkitc-1/LovyanGFX/src/lgfx_user/LGFX_ESP32S3_RGB_ESP32-8048S043.h"
#include "lgfx_user/LGFX_ESP32S3_RGB_ESP32-8048S043.h"
#include <LGFX_TFT_eSPI.hpp>

// #include "ESPNowCam.h"
#define CORE_DEBUG_LEVEL = 3

static LGFX tft;

// ESPNowCam radio;

// frame buffer
// uint8_t *fb;

// display globals
int32_t dw = 800;
int32_t dh = 480;

// void onDataReady(uint32_t lenght) { tft.drawJpg(fb, lenght, 0, 0, dw, dh); 
}

void setup(void) {
  delay (10000);
  Serial.println("begin setup");
  tft.init();
  tft.setRotation(0);

  pinMode(2, OUTPUT);
  analogWriteResolution(8);    // PWM resolution
  analogWriteFrequency(1000);  // PWM frequency
  analogWrite(2, 127);         // Brightness set

  tft.setTextColor(TFT_WHITE);
  tft.fillScreen (TFT_RED);
  delay (10000);
  // BE CAREFUL WITH IT, IF JPG LEVEL CHANGES, INCREASE IT
  /*
   * fb = static_cast<uint8_t*>(malloc(30000 * sizeof(uint8_t)));

   * radio.setRecvBuffer(fb);
   * radio.setRecvCallback(onDataReady);

   */
  // if (radio.init()) {
    String text = "ESPNow Init Success";
    uint8_t textLenght = text.length();
    uint8_t textSize = 4;
    uint16_t posX = (800 - (textSize * 6 * textLenght)) / 2;
    uint16_t posY = ((480 - (textSize * 8)) / 3) * 2;
    tft.setTextSize(textSize);
    tft.drawString(text, posX, posY);
  // }
  delay(500);
}

void loop(void) {}
```

Extra clue: when connecting the board to Arduino IDE and selecting board 
ESP32S3 Dev LModule, it is recognized by IDE and and appears with thes 
informations:

Flash Mode:  "QIO 80MHz"
Flash Size: "4MB (32Mb)"
Partition scheme: "Default 4MB with spiffs (1.2MB APP/1?5MB SPIFFS)"

I tried to translate these informations in platformio.ini file with these new 
lines:

board_build.arduino.memory_type = qio-opi
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_upload.flash _size = 4MB

but il led to the same failure
At last, changing board_build.arduino.memory_type  from qio-opi to opi did not 
change anything.
PSRAM: "Disabled"

phpo1-61, 2025-01-29:
Hie,

You will find hereafter a more readable description of my tests.
I made a preview on Dillinger website and performed an export in md format.

Looking forward for you advice.

Best regards.

Philippe Poutrel.

phpo1-61, 2025-01-30:
As receiver of ESPNowCam protocol flow, I am testing the ESP32-8048S043 board 
(ESP32 S3 WROOM-1 with capacity touch 4,3 inch 800 X 400 LCD TFT RGB screen, 
ILI9485 driver according to the Epressif datasheet for this processor). I 
choosed this model because il matches the driver file parameters named 
LGFX_ESP32S3_RGB_ESP32-8048S043.h in github ESPNowcam website.

I suppose that this processor is not fitted with internal PSRAM, otherwise its 
denomination would involve the "R2" or "R8" pattern and no other clue shows 
that a PSRAM is included on the board. It seems to be confirmed by the fact 
that the board used in the tft-rgb-basic-receiver.cpp example file (github 
ESPNowCam website) is esp32-s3-devkitc-1 and appears with the attributes 
N8(8MB-QD, No PSRAM) in PLatformIO when choosing a board, new project creation 
process.

My work environment is:

```python
SCode version: 1.96.0
Date: 2024-12-11 1T02:29:09.626Z(2wks ago)
Version: 1.96.0
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Linux x64 6.12.4-200.fc41.x86_64

platformIO version:

***@***.***:~/Documents/PlatformIO/Projects/test1$ pio system info
--------------------------  --------------------------------------------------
PlatformIO Core             6.1.16
Python                      3.13.0-final.0
System Type                 linux_x86_64
Platform                    Linux-6.12.4-200.fc41.x86_64-x86_64-with-glibc2.40
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/admin/.platformio
PlatformIO Core Executable  /home/admin/.platformio/penv/bin/platformio
Python Executable           /home/admin/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10

I use the Arduino platform within PLatformIO, not ESP-IDF.
```

First, I made a simple configuration in order to test the display function 
based on the tft-rgb-basic-receiver.cpp and platformio.ini just to check the 
completion of basic Serial.print  and  fillscreen TFT_RED commands.

In main.cpp file:
- all commands dealing with ESPNoCam library have been commented
- #DEFINE CORE_DEBUG-LEVEL =3  has been added (in order to get detailed ESP32 
exceptions messages)

In platformIO.ini file:
 - board_build.flash_size = 16MB command has been used, although I am not sure 
of this size
 - -D BOARD_HAS_PSRAM has been commented, as suggested previously
 - only Lovyan03 library is loaded in working environment

 platformio.ini file:

```python
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
[... 267 lines cut ...]
PSRAM chip not found or bad type or wrong PSRAM line mode.

If at last PSRAM actually exists on this board, how could I declare it with 
DIO mode in platformio.ini file (this information is given in Exception 
messages) since I understood that FLASH and PSARM operates the same SPI 
transfert format mode (SPI, Dual SPI, Quad SPI, OctalSPI, QPI, OPI)

 - Without D BOARD_HAS_PSRAM, with board_build.flash_size = 16MB, without 
specifying partitions.csv, 1.2.0 Lovyan03 library release loaded: same 
behavior and exception messages

 - with additional commands in platformio.ini file (found among the board 
details when connected to Arduino IDE) and without PSRAM

 ```python
 board_build.arduino.memory_type = qio_pio
 board_build.f_flash = 80000000L
 board_build.flash_mode = qio
 board_uoload.flash_size = 4MB
 ```

I would like to have your advice in order to investigate about the origin of 
this issue because I'm now running out of ideas.

Best regards.

phpo1-61, 2025-01-31:
As receiver of ESPNowCam protocol flow, I am testing the ESP32-8048S043 board (ESP32 S3 WROOM-1 with capacity touch 4,3 inch 800 X 480 LCD TFT RGB screen, ILI9485 driver according to the Epressif datasheet for this processor). I choosed this model because il matches the driver file parameters named LGFX_ESP32S3_RGB_ESP32-8048S043.h in github ESPNowcam website.

My work environment is:

```python
SCode version: 1.96.0
Date: 2024-12-11 1T02:29:09.626Z(2wks ago)
Version: 1.96.0
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Linux x64 6.12.4-200.fc41.x86_64

platformIO version:

admin@pc-29:~/Documents/PlatformIO/Projects/test1$ pio system info
--------------------------  --------------------------------------------------
PlatformIO Core             6.1.16
Python                      3.13.0-final.0
System Type                 linux_x86_64
Platform                    Linux-6.12.4-200.fc41.x86_64-x86_64-with-glibc2.40
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/admin/.platformio
PlatformIO Core Executable  /home/admin/.platformio/penv/bin/platformio
Python Executable           /home/admin/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10

I use the Arduino platform within PLatformIO, not ESP-IDF.
```

First, I made a simple configuration in order to test the display function based on the tft-rgb-basic-receiver.cpp and platformio.ini just to check the completion of basic Serial.print  and  fillscreen TFT_RED commands.

In main.cpp file:
- all commands dealing with ESPNoCam library have been commented
- #DEFINE CORE_DEBUG-LEVEL =3  has been added (in order to get detailed ESP32 exceptions messages)

In platformIO.ini file:
 - board_build.flash_size = 16MB command has been used, although I am not sure of this size
 - -D BOARD_HAS_PSRAM has been commented
 - only Lovyan03 library is validated in working environment (hpsaturn has been loaded, but line is commented)

 platformio.ini file:

```python
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.flash_size = 16MB
board_build.partition = "home/admin/Documents/PlatformIO/Projects/test14/.pio/libdeps/esp32-s3-devkitc-1/EspNowCam/config/partitions.csv"
monitor_speed = 115200
monitor_filters =
  esp32_exception_decoder
  time
[... 176 lines cut ...]
At last, I achieved 2 extra tests:

  1 - with D BOARD_HAS_PSRAM in platformio.ini and 1.2.0 release (Lovyan03): compilation fails with this message:

 E(109) PSRAM ID read error: 0X00ffffff, PSRAM chip not found or no supported,or wrong line mode

 2 - According to its description (but not mentionned on the ESP32 chip), this board seems to be fitted with 16MB flash and 8MB PSRAM.

Having carefully red the ESP32S3-WROOM-1 datasheet, I learned that this kind of chip (N16R8) uses octal SPI for PSRAM, Quad SPI for Flash and works at 80Mhz frequency for flash mode.

SO I tried to take in account these specifications with the platformio.ini following extra commmands:

 ```python
 board_build.arduino.memory_type = qio-opi
 board_build.f_flash = 80000000L
 board_build.flash_mode = qio
 board_build.flash_size = 16MB
 ```

But always the same issue and same core exception recurrent message...

I would like to have your advice in order to investigate about the origin of this issue because I'm now running out of ideas.
Is it a good idea the pPlatformIO platform parameter from Arduino to ESP-IDF (not yey tested)?

Best regards.

phpo1-61, 2025-01-31:
Le vendredi 31 janvier 2025, 12:16:59 heure normale d’Europe centrale hpsaturn 
a écrit :
> Hi, please enter via GitHub interface, and please review your posts, because
> your code blocks are not working. Maybe do you answer it via Email? if yes,
> you are wrong. This is not the way. On the other hand, you have a problem
> with your stuff using Arduino IDE. Please use PlaformIO, because it is
> full-replicable and we can talk the same language.
Hi, sorry for my previous misformatted messages.
I've just answered in github, and the preview ouput seems to bet better: I 'm 
improving github and hope it with be so with ESP32S3-WROOM-1 !

Best regards.

Philippe Poutrel.

hpsaturn, 2025-01-31:
Thanks to fix the format issues.
Well, I don't have this panel, but I can see some things:

1. You have commented some important lines on the Pio ini file. Please, keep in mind that the ; is a comment in this kind of files
2. Can you please share the header o config file for LovanGFX that you are using?
3. Sounds that is a problem with the initialization of the TFT.
4. Are you already tested a Hello World of this TFT panel using LGFX and PlatformIO? If yes, can you please share a GitHub link of this sample or please attach a zip of the complete example project.

hpsaturn, 2025-02-01:
The user @MeloCuentan reports to me that he has a similar board like your board, working, with the next settings:

```python
[env:lolin_s3]
platform = espressif32
board = lolin_s3
framework = arduino
monitor_speed = 115200
lib_deps = 
  lovyan03/LovyanGFX@^1.1.12
  hpsaturn/EspNowCam@^0.1.13
```

```cpp

#define LGFX_USE_V1
#include <LovyanGFX.hpp>

#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>

#include <driver/i2c.h>

class LGFX : public lgfx::LGFX_Device
{
public:
  lgfx::Bus_RGB _bus_instance;
  lgfx::Panel_RGB _panel_instance;
  lgfx::Light_PWM _light_instance;
  lgfx::Touch_GT911 _touch_instance;

  LGFX(void)
  {
    {
      auto cfg = _panel_instance.config();

      cfg.memory_width = 800;
      cfg.memory_height = 480;
      cfg.panel_width = 800;
      cfg.panel_height = 480;

      cfg.offset_x = 0;
      cfg.offset_y = 0;

      _panel_instance.config(cfg);
    }

    {
      auto cfg = _panel_instance.config_detail();

      cfg.use_psram = 1;

      _panel_instance.config_detail(cfg);
    }

    {
      auto cfg = _bus_instance.config();
      cfg.panel = &_panel_instance;
      cfg.pin_d0 = GPIO_NUM_8;   // B0
      cfg.pin_d1 = GPIO_NUM_3;   // B1
      cfg.pin_d2 = GPIO_NUM_46;  // B2
      cfg.pin_d3 = GPIO_NUM_9;   // B3
      cfg.pin_d4 = GPIO_NUM_1;   // B4
      cfg.pin_d5 = GPIO_NUM_5;   // G0
      cfg.pin_d6 = GPIO_NUM_6;   // G1
      cfg.pin_d7 = GPIO_NUM_7;   // G2
      cfg.pin_d8 = GPIO_NUM_15;  // G3
      cfg.pin_d9 = GPIO_NUM_16;  // G4
      cfg.pin_d10 = GPIO_NUM_4;  // G5
      cfg.pin_d11 = GPIO_NUM_45; // R0
      cfg.pin_d12 = GPIO_NUM_48; // R1
      cfg.pin_d13 = GPIO_NUM_47; // R2
      cfg.pin_d14 = GPIO_NUM_21; // R3
      cfg.pin_d15 = GPIO_NUM_14; // R4

      cfg.pin_henable = GPIO_NUM_40;
      cfg.pin_vsync = GPIO_NUM_41;
      cfg.pin_hsync = GPIO_NUM_39;
      cfg.pin_pclk = GPIO_NUM_42;
      cfg.freq_write = 14000000;

      cfg.hsync_polarity = 0;
      cfg.hsync_front_porch = 8;
      cfg.hsync_pulse_width = 4;
      cfg.hsync_back_porch = 16;
      cfg.vsync_polarity = 0;
      cfg.vsync_front_porch = 4;
      cfg.vsync_pulse_width = 4;
      cfg.vsync_back_porch = 4;
      cfg.pclk_idle_high = 1;
      _bus_instance.config(cfg);
    }
    _panel_instance.setBus(&_bus_instance);

    {
      auto cfg = _light_instance.config();
      cfg.pin_bl = GPIO_NUM_2;
      _light_instance.config(cfg);
    }
    _panel_instance.light(&_light_instance);

    {
      auto cfg = _touch_instance.config();
      cfg.x_min = 0;
      cfg.x_max = 800;
      cfg.y_min = 0;
      cfg.y_max = 480;
      cfg.pin_int = GPIO_NUM_18;
      cfg.bus_shared = false;
      cfg.offset_rotation = 0;
      // conexión I2C
      cfg.i2c_port = I2C_NUM_1;
      cfg.pin_sda = GPIO_NUM_19;
      cfg.pin_scl = GPIO_NUM_20;
      cfg.freq = 400000;
      cfg.i2c_addr = 0x14; // 0x5D , 0x14
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);
    }

    setPanel(&_panel_instance);
  }
};
```

phpo1-61, 2025-02-02:
Cheers, it works!!!
Thank you for your help.
I notice that changes consists in:

 - type of selected board: WEMOS LOLIN S3 (named lolin_s3 in platformio.ini) instead of Espressif esp32-s3-devkit-c1
 - add the declaration of pin_brightness in main.cpp

Last remark: the video image only covers the upper left corner of the board's screen (less than one quarter of the surface), though the tft.drawJpg inside the onDataReady function includes actual width and height of the screen (800 and 480). How can I fix this restriction?

And would it be useful to other users (newbies like me in EP32 matter) if i published simple platformio.ini and main.cpp files for my configuration (one for the ESP32Cam AiThinker emitter board, one for the Lilygo TTGo T-Display receiver board and one for teh ESP32-8048S043 receiver board? That's to say one Platformio project per board.

In fact, first  I spent some time to understand how your general  platformio.ini file was organized in order to retrieve useful informations about my boards  and then I had some trouble to find the path to driver.h file.
 Including driver.h file content in main.cpp file is easier to understand, though less clever than #include

Best regards.

hpsaturn, 2025-02-03:
I'm glad to hear that is working. Please the other questions maybe could be in the [Telegram ESP32 group](https://t.me/esp_es) that we have with other hobbyists. You able to ask in English or Spanish. For now, I'm going to close this issue.

[image]

closed 2025-02-03 by hpsaturn