hpsaturn/ESPNowCam #57: not an issue but an apparent success for AtomS3R camera module

Mich2e, 2025-02-04:
I've tested the camera part of your project with a ATOMS3R module and it seems to work fine.
average FPS with your setting is 9-11 fps
I will esplore a bit more, as I'm not using your example for receicer: my attempt is to view on a phone, using an espS3 connected on USB.
Congrats and thanks for that !

Mich2e, 2025-02-04:
I would be pleased to do so, and share what I do, but please be aware I'm not a technician, and most often I do not understand what I'm trying to do. What do you mean by sample ? do you mean the script I used ? For the receiver I'm facing many - basic ?? - issues. cannot return to working version :[

hpsaturn, 2025-02-04:
Don't worries. Just I was say that when you reach a full example, that maybe could help to support and expand the ESPNowCam library examples, and maybe, with that, you could submit a pull request to this repo to contribute and grow the library :D

Mich2e, 2025-02-04:
Huumm I'm starting to understand that I cannot join file here :-) one of the reasons for pull request. here is link to current set up parly working: the emitter with an AtomS3R....emits, the receiver based on a S3Stamp is receiving and serving a web page. so far display is up/down and with sometimes few colored bars, but that 's fair for intended use: remote liveview. did not yet experimented outdoor https://e.pcloud.link/publink/show?code=XZQRCKZcUNi5qObRi0fiUt6Og6pBjsChitk
Note I had a look to pull request. unfortunately not clear for me how to handle this. so for the first attemps this will be ok.

hpsaturn, 2025-02-04:
sorry, I don't have access to this cloud. But you can publish this in GitHub, for instance in a repo of your tests. Also here, in this thread, you can add attachments. And also here, in each comment, you can put block of codes formatted in markdown. It is very easy, using triple backticks:
https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

Mich2e, 2025-02-12:
I hope this link is ok now ?
here is ongoing project with AtomS3R. https://github.com/Mich2e/EspNowLiveViewTest 
I've added a option to send a couple of commands from the receiver. note that speed, at least for test is quite good. I've set the speed at 10 fps and this is ok.

hpsaturn, 2025-02-13:
Sorry, your repository is empty. Maybe you forget to send the files, using `git push`

hpsaturn, 2025-02-13:
Don't worries step to step. Like in my previous message. You also can add files here in the comments. On the other hand, these are some guides:

https://product.hubspot.com/blog/git-and-github-tutorial-for-beginners
https://docs.github.com/en/get-started/start-your-journey/hello-world
https://www.freecodecamp.org/news/introduction-to-git-and-github/

hpsaturn, 2025-02-14:
Yes, thanks. I notice that OKtestFinalLVNow.ino is a web server that receive the frames, and then it will shows these via the web server. I think for view the frames in the mobile. is it true?

Nice idea. I didn't test it yet. But sounds very interesting to add it like a example. Maybe do you have a video demo? that will be nice.

Mich2e, 2025-02-14:
Yes you're right: at the end target is to view images on the phone and check if range is longer than simple websocket between esp and phone directly.
for video, actually, I'm not so fast !! as you may have noticed. 
the only thing so far as test is a very (!) small clip that shows the idea. you can see it there. https://e.pcloud.link/publink/show?code=XZiwvKZY6IU2tx7FEbnilRWdeVaCm0Tyg97 It's only a kind of "proof of concept". more focussed on sending infos from the phone to the esp, than really video part of it. but it is working. 
plan now is to test it outdoor to check real acceptable range. but to do so there is still a list of roadblocks: power supply is one of them (not sure 3.6 battery will last enough) and moreover the support for all this stuff ! Another trick is that for unknown reason the  wifi AP for server is not starting if the receiver is directly powered by phone.... 
today I've tried to implement also the imu datas from the espcam....so far without success. I've got a compilation error, and don't know how to deal with. :(

hpsaturn, 2025-02-17:
Right now I'm a little busy, but I want to replicate your idea, because I notice that is possible improve it, for instance using the EspNowCam callbacks in the both side without using the original EspNow callback, I think. Nevertheless, thanks for share this proof of concept, it's a good idea 😃

Mich2e, 2025-03-06:
Hi, just to follow up on this and share that the option to use a small atom to display is also tested and works https://github.com/Mich2e/EspNowLiveViewTest/blob/main/rec_Atom_test.ino
Not sure this is a "great" achievement with this tiny screen ;-) but the sketch is still compliant with display on connected phone.