hpsaturn/ESPNowCam #58: Limitation of ESPNowCam with an 800 X 480 screen

phpo1-61, 2025-02-26:
I tested ESPNowCam with the ESP32Cam AIThinker and 2 receivers boards: LILYGO T-Display (240 X 180 TFT screen) and ESP32-8048S043C_I (800 X 480 RGB screen). Both receivers operate, however the last one is restrictive as the camera's picture only covers the uppper left corner of the screen.

You will find each configuration's board in the following GihHub repository:

https://github/com/phpo1-61/My-tests-with-ESPNowCam.git

To try to find the origin of the issue, I tried in vain to increase the resolution of the camera with the leading changes in CamAIThinker.h file

xclk_freq_hz changed from 16 000 000 to 20 000 000:

config_jpeg_quality changed from 12 to 6

config_frame_size changed from  FRAMESIZE_QVGA to FRAMESIZE_VGA, andd then FRAMESIZE_SVGA

Thank you for your advice.

Best regards.

Philippe POutrel.

hpsaturn, 2025-02-26:
I'm trying to understand your problem. After you changing the frame_size like this:

```cpp 
  Camera.config.frame_size = FRAMESIZE_SVGA; // or VGA
  if (!Camera.begin()) {
    Serial.println("Camera Init Fail");
  }
```
do you have the same problem? I mean, the problem is that the image is small in the screen receiver?

phpo1-61, 2025-02-27:
Le mercredi 26 février 2025, 22:23:21 heure normale d’Europe centrale hpsaturn 
a écrit :
> hpsaturn left a comment (hpsaturn/ESPNowCam#58)
> 
> I'm trying to understand your problem. After you changing the frame_size
> like this:
> 
> ```cpp
>   Camera.config.frame_size = FRAMESIZE_SVGA; // or VGA
>   if (!Camera.begin()) {
>     Serial.println("Camera Init Fail");
>   }
> ```
> do you have the same problem? I mean, the problem is that the image is small
> in the screen receiver?
Hello, You've fixed the issue: now with VGA resolution, the picture covers the 
exact VGA resolution on the screen: 640 X 480. However and as you mentionned 
previously, increasing the camera resolution generates a higher network 
troughput  and so a decreased fps rate on the screen. Now I am able to adjust 
between picture resolution and display refresh rate. Thanks a lot!

In fact, I understand that adjusting the camera's parameters must be done 
after its initization and not before as  I tried.  

I will continue to follow your github site and its associated issues. Thanks 
to you, I became familliar with markdown langage and git & GitHub which are 
really efficient in order to share informations and knowledge.

Best regards.

Philippe Poutrel.

hpsaturn, 2025-02-28:
Ok, good, for now I'm going to close this issue. On the other hand, I'm working in the last item of my TODO list:

- [ ] Migration to esp_wifi_80211_tx() to improve Payload and Quality

because, in the last review of my library by the people of Linux Embedded, they did a mention this possibility too:

https://www.cnx-software.com/2025/02/24/espnowcam-library-enables-esp32-video-camera-or-data-transmission-with-the-esp-now-protocol/

and I want return again to this idea.

closed 2025-02-28 by hpsaturn

phpo1-61, 2025-03-09:
Le mercredi 26 février 2025, 22:23:21 heure normale d’Europe centrale hpsaturn 
a écrit :
> hpsaturn left a comment (hpsaturn/ESPNowCam#58)
>
> I'm trying to understand your problem. After you changing the frame_size
> like this:
>
> ```cpp
>   Camera.config.frame_size = FRAMESIZE_SVGA; // or VGA [...]
Hello,
I have enhanced my GitHub site:

 https://github.com/phpo1-61/My-tests-with-ESPNowCam/blob/main/README.md

with a READ.ME file in order to detail my tests with ESPNowCam.

Thanks one more time for your help.