hpsaturn/ESPNowCam #59: Add capability to set WiFi Channel

RickLloyd24, 2025-03-08:
The library is great and easy to use.  You need to add the capability to set the WiFi Channel.  My receiver board must connect to the WiFi network and receive ESP-NOW messages.  When the router resets in changes the WiFi channel.  It normally picks channel 1 or 6.  When channel 6 is used, I can't communicate between the sender and the receiver.

hpsaturn, 2025-03-09:
Thank you for your positive comments. Are you already have a way to handle the channel? I never tested this, but I did a first approach, please try to test if this could help:

https://github.com/hpsaturn/ESPNowCam/pull/60

You are able to test it using this branch in the PR like a PlatformIO dependency with target this Github and its branch with #, like this:

```python
lib_deps = 
  https://github.com/hpsaturn/ESPNowCam.git#wifi_channel_issue_59
```

RickLloyd24, 2025-03-10:
I am using Arduino so I can't try your suggestion.  Since your library did not have the capability to set the channel, I used the standard #include <esp_now.h> library.  I wrote the issue so that other people that have Wi-Fi on different channels can use your library.  Adding the capability to set the channel in the radio.init(); function would be a great improvement.

I also use an auto pairing routine to determine the Wi-Fi channel. There is an article that explains the process in detail at the following link: https://randomnerdtutorials.com/esp-now-auto-pairing-esp32-esp8266/.

I my applications, the sender stores data in PSRAM and then sends the stored data.  Unlike most applications I cannot accept bad packets. Normally I am getting response times of about 166 milliseconds or 6 frames per second.  I had to use the callback feature (esp_now_register_send_cb(OnDataSent) to verify that the 245-byte packet was successfully received.  I found that depending upon network traffic in my house my transmit times change.  I have seen the time go up to 400 milliseconds or 2.5 frames per second.  I also had to implement a re-try routine.  When the response status is "ESP_NOW_SEND_FAIL", I send the packet again.  So far, I have never seen the second packet fail.  In the packet the data is 244 bytes, and one byte is a retry flag.

#60<https://github.com/hpsaturn/ESPNowCam/pull/60>

lib_deps =

  https://github.com/hpsaturn/ESPNowCam.git#wifi_channel_issue_59

-

#60<https://github.com/hpsaturn/ESPNowCam/pull/60>

lib_deps =

  https://github.com/hpsaturn/ESPNowCam.git#wifi_channel_issue_59

-

hpsaturn, 2025-03-11:
Thanks for your response. Some things, first, please respond to these messages using the GitHub webpage, please not use the Email way, because it put a lot of unnecessary info in this thread.

If you are using Arduino IDE, you are able to test this new proposal using its branch and downloading the "new library test" from here:

https://github.com/hpsaturn/ESPNowCam/tree/wifi_channel_issue_59 (using Code button, Download zip)

remove the previous library and add this zip contents to your Arduino IDE library directory.

After that you have a new library setter, you will able to use something like this:

```cpp
radio.setChannel(6);
radio.init();
```

please let me know if this new library works or resolve the issue.

hpsaturn, 2025-04-06:
Released and tested this new feature on the last version v0.1.17. Please let me know if this resolve the issue. Thanks.

closed 2025-04-06 by hpsaturn