hpsaturn/ESPNowCam #63: Not an issue but FREENOVE-WROVER works

E1100, 2025-05-17:
I found that my ESP32-WROVER-CAM from Freenove supports this library from the "generic-camera-psram" sketch with the configured pins. Runs at around 9 fps. Unfortunately, I can't get video buffer because the other side crashes when getting buffer with the LoadProhibited error. (I am using ESP32-WROOM, I don't know if that is supported for receiving)

hpsaturn, 2025-05-19:
In theory it should works in any receiver. Maybe is the allocation of the buffer in your receiver. Can you share some code about your implementation?

E1100, 2025-05-21:
> In theory it should works in any receiver. Maybe is the allocation of the buffer in your receiver. Can you share some code about your implementation?

I'm a beginner so my code probably was the reason why it crashed in the first place. I edited the "tft-3.5-basic-receiver" sketch and changed the output, but I think my way of allocating the buffer was bad. Unfortunately, I have lost the sketch, but can you make some example code for me to test?

hpsaturn, 2025-05-22:
If you want help with that, you need share your code.  If have more problems please open a new issue with the details. For now I'm closing this, but feel free to open more issues for help or support you using this library.

closed 2025-05-22 by hpsaturn